OpenChrom/openchrom #1: GeneIdent is missing

Mailaender, 2019-09-02:
and the original repository https://github.com/OpenChrom/GeneIdent is archived so I can't access the issues as well.

eselmeister, 2019-09-02:
We moved most of the Repos to:
https://github.com/OpenChrom/openchrom

I'll inspect it. If GeneIdent is missing, I'll merge it soon into the 
"openchrom" main repo.

Best,
Philip

eselmeister, 2019-09-02:
Code from the archived repository has been merged. We are working constantly to make the OpenChrom development more comprehensive and easier. Have a look here how to contribute:
https://github.com/OpenChrom/openchrom/blob/develop/CONTRIBUTING.md

closed 2019-09-02 by eselmeister

Mailaender, 2019-09-21:
Seeing how incomplete and now since the data analysis view merge effectively broken the GeneIdent prototype is, it might be better to keep it in its previous separate repositories. The issues also seem out of context here.