OpenChrom/openchrom #11: Targets View remains incomplete

danield5732, 2019-09-04:
After peak identification via NIST14, entries in the Targets View does not contain the Mol Weight which would be usefull with respect to the parent ion.

Also in some cases the CAS number is 0-00-0. 
I guess this leads to no creation of SMILES code and therefore no molecule structure is shown.
Examples are: 

- Heneicosyl trifluoroacetate
- E-15-Heptadecenal
- Oxime-, methoxy-phenyl-_

danield5732, 2020-06-04:
Update: 
CAS number is 0-00-0, if it is missing in the NIST database, too, so this is ok.
SMILES code was related to Issue #40 and therefore is fixed. Anyhow, some SMILES codes are not calculated by Chromatogram Identifier - SMILES Identifier (e.g. Iminostilbene and Chamazulene).

For this, the only remaining issue is that the Mol Weight collumn is not filled. Also when using NIST17.

Mailaender, 2021-03-22:
[image]

seems to be parsing .msp properly so I guess missing mol weight is also missing in the database which then has to be calculated with a chromatogram identifier.

closed 2021-03-23 by eselmeister