OpenChrom/openchrom #7: Zoom in Comparison Scan mode cuts off masses

danield5732, 2019-09-04:
Zooming into m/z specta in the Comparison Scan Mode window may lead to loss of single mass bars.
It occures when the zoom does not include the maximum value of the mass bar. Then, the m/z value does not exist in the zoomed view anymore.

eselmeister, 2019-09-04:
Thanks for reporting. Are you able to share some screenshots, showing 
the problem?

danield5732, 2019-09-04:
The complete spectra:
[image]

After zoom, m/z 193.2 disappears
[image]

eselmeister, 2019-09-05:
The series compress setting was set to true, which causes the m/z values to disappear. It has been fixed. Compress is set to false by default for the scan chart now. But one can toggle the compress option in the settings.

[image]

closed 2019-09-05 by eselmeister