OpenChrom/openchrom #9: SMILES is wrong calculated for Iminostilbene

danield5732, 2019-09-04:
This leads to a wrong molecular structure in the Molecule View window.

[image]

The correct one would be: C1=CC=C2C(=C1)C=CC3=CC=CC=C3N2

eselmeister, 2019-09-05:
We are using OPSIN to parse the names:
https://opsin.ch.cam.ac.uk

Probably, a bug needs to be filed here:
https://bitbucket.org/dan2097/opsin/src/default/

[image]

closed 2019-09-05 by eselmeister

eselmeister, 2019-09-05:
The used OPSIN library has been updated to version 2.4.0. A warning is generated as shown in the screenshot. Calculating the SMILES is skipped now, if a warning exists.