OpenChrom/openchrom #15: Comparison Scan Modus Window blocks right click in tables

danield5732, 2019-09-18:
If the Window for the comparison scan modus is activated it is refreshed every time when clicking in the peak list table or the targets table.
In case of right click in these tables the context menu disappears due to refreshing the comparison scan modus window.
Therefore, the context menu of right clicks are not available.

eselmeister, 2019-09-19:
Could you supply a screenshot?

danield5732, 2019-09-19:
Unfortunately not, since the context menu shows up very shortly and disappears again.
Talking about the right click context menu for the peak list table and the targets table.

laeubi, 2019-11-19:
This is now resolved by https://github.com/eclipse/chemclipse/issues/116

closed 2019-11-19 by laeubi