OpenChrom/openchrom #19: Identifier NIST-DB issue

judrosden, 2019-10-11:
Everything appears to run OK connecting with the NIST DB as I have highlighted peaks in my peak list and run the identifier command. I get a message with a green status bar that it may take a few minutes. No errors are thrown. However the targets view doesn't get populated with any compound ids.

Also it looks like some .HLM files are generated in the NIST directory when I run the program.

[image]

[image]

[image]

I'm using OpenChrom Community Edition   OpenChrom Community Edition (Dalton)	1.3.0.201808081331

And NIST version 2.3

Any ideas what I am missing?

kerner1000, 2019-10-12:
Hi judrosden, did you try to re-select the chromatogram to trigger a refresh of the peak list?

judrosden, 2019-10-17:
Hello, Yes, that doesn't seem to make a difference.

Judith Denery
Head of Analytics at Cana
judith@cana.com (510) 473-7317

laeubi, 2020-01-08:
@judrosden 1.3 is rather old, the upcoming 1.4 release contains several-bugfixes regarding NIST-Integration, can you please try out if you issue is solved there?

You can download it here: https://lablicate.com/platform/openchrom/download

BTW: If you are using NIST-DB at daily work we also offer a more advanced NIST-Integration, it is currently under development but will be available soon as a payed extension in our marketplace, let me know if you are interested in participate in the Beta-Test.

kerner1000, 2020-02-06:
I assume this is resolved. There is also a wiki entry for nist peak identifier now.

closed 2020-02-06 by kerner1000