OpenChrom/openchrom #20: Missing requirement: CDK Support 1.4.0.201909170105

Mailaender, 2019-10-16:
I tried to export the OpenChrom .product file and I get:

```
 Cannot complete the install because one or more required items could not be found.
 Software being installed: OpenChrom Analytics Edition (Lawrence) 1.4.0.201910161021 (net.openchrom.rcp.compilation.community.product.id 1.4.0.201910161021)
 Missing requirement: CDK Support 1.4.0.201909170105 (net.openchrom.chromatogram.msd.identifier.supplier.cdk.feature.feature.group 1.4.0.201909170105) requires 'net.openchrom.thirdpartylibraries.cdk.feature.feature.group 0.0.0' but it could not be found
 Cannot satisfy dependency:
 Cannot satisfy dependency:
```

laeubi, 2019-10-16:
Sadly the product export wizzard is a bit odd sometimes, thats why we recommend building the product on commandline see https://github.com/OpenChrom/openchrom/wiki/Build

laeubi, 2019-12-04:
@Mailaender Did this resolves your issue?

Mailaender, 2019-12-04:
I am currently using Windows at work so the easiest way was to stop updating the whole OpenChrom and instead export only the features and plugins I changed to replace them manually on each PC.

laeubi, 2020-01-08:
Okay I'll mark this as resolved then, feel free to open a new ticket if the issue still persists.

closed 2020-01-08 by laeubi