OpenChrom/openchrom #23: Missing Reference Chromatogram for *.D Chromatogram

kerner1000, 2019-11-21:
When opening a Chromatogram directly from the .D folder, I do not have the option to select a reference chromatogram.

[image]

If I save this chromatogram to ocb and reload this file, I can choose to display a reference chromatogram.

[image]

eselmeister, 2019-11-21:
> no-reference

I can't reproduce this issue.

When opening chromatograms, the responsible import takes care to load 
additional/related chromatograms on demand. Each converter may list some 
settings to allow loading accompanying data, e.g. when loading a MSD 
file, load also the FID file, which was measured in split modus.

If no reference has been loaded, no reference will be listed in the 
dropdown box. It's not a matter of the file format.
Alternatively, you could use the green plus button to add new 
reference(s) manually.

kerner1000, 2019-11-21:
Cannot reproduce on latest develop.

closed 2019-11-21 by kerner1000