OpenChrom/openchrom #24: Why I can not  run openchrom in eclipse ？

IMAUZS, 2019-11-30:
I follow the QuickStart in developer manual，but  there's still appear many compile error  .
Does any where I got wrong about the configuration？
@laeubi

laeubi, 2019-11-30:
What compile error do you get?

IMAUZS, 2019-11-30:
I think console appears many errors because of load target platform failed.
In my eclipse automatically download  target  platform is very slow. I don't know if load target platform uncompletely is the reason of appearing many errors.

eselmeister, 2019-11-30:
This could be a reason. Would you like to run OpenChrom inside of a specific Eclipse IDE or inside of an Eclipse product? If yes, which?Best,Philip

IMAUZS, 2019-11-30:
I use the eclipse-rcp as you recommend that is eclipse-rcp-oxygen-3a-win32-x86_64,and I download the eclipse-rcp in eclipse website.

IMAUZS, 2019-11-30:
Sir, the errors as the bellow pictures.

IMAUZS, 2019-12-03:
Hello,sir,I already compile the OpenChrome successfully by eclipse.

IMAUZS, 2019-12-03:
Hello,sir,I already compile the OpenChrome successfully by eclipse.
Now I have a question that how can i localize the software to Chinese?
I don't see the menu that can re-localize the software view language.
@laeubi@eselmeister

eselmeister, 2019-12-04:
We basically have the support to enable i18n and l10n. Currently, the 
default for I/O and UI is english. Translation requires a lot of work as 
all items of the software need to be adjusted. Anyways, it's planned for 
the next year. Are you willing to support us?

Best,
Philip

IMAUZS, 2019-12-04:
Maybe I can provide some helps to make this.I want to use OpenChrom to control my device and analyze the chromatogram signal.It's difficult to develop a new software frow zero.So If  OpenChrom can localize to Chinese, then I can use OpenChrom directly. It will be great for me and everyone.
I'm very glad to  join you to raise the OpenChrom.I'm a Chinese software engineer,so I can help to translate the text to chinese.If you need me,you can email to me 351092161@qq.com.

Best Regards,
zhangshuai

eselmeister, 2019-12-04:
Which kind of instrument would you like to run/control?

laeubi, 2019-12-04:
I think your best choice is to live with the English version for now and wait until we have connected chemclipse (whats the base for openchrom) to the Eclipse-Babel project, that has several advantages:

1. Many of the standard JFace-Widgets are already translated to several languages
2.It is very easy to contribute new translations even by non software developer

But this will be more a long running story than a thing that will happen in the next weeks.

IMAUZS, 2019-12-04:
@eselmeister 
Our instrument is mainly used for separating and purifying impurities.The instrument is a new product in our company.So we want to find a software to control instrument  and analyze the impurity signals.

IMAUZS, 2019-12-04:
@laeubi 
You advice me use chemclipse firstly and thers's already plugin can translate to other languages in chemclipse .Is that your meaning?I just open chemclipse,but I can not search the translation plugin in marketplace .
Now our company client is chinese,so we need the software interface is chinese text.

laeubi, 2019-12-04:
@IMAUZS OpenChrom uses ChemClipse as its platform so we should start there but this feature is currently not enabled, it is maynly driven by opensource-enthusiasts and volunteer work.

For me it reads as if you are developing software for a commercial product, so you might be interested in the commercial-support for Openchrom see https://lablicate.com/services for details.

With commercial support you can get:

- Get specific help with integration of your instrument/data with OpenChrom
- Custom extensions for advanced integration of specific products or workflows
- We can even offer you a support plan that includes preparing the software for translation of the software in chinese, so you only need to supply the required translation strings.

IMAUZS, 2019-12-04:
@laeubi Ok,Thanks for your explains.I will have a look what you have mentioned.When I need more commercial-support ,I will send message to you .

eselmeister, 2019-12-09:
Is it a GC/MS or GC/FID from a known vendor like Shimadzu, Agilent?

IMAUZS, 2019-12-09:
Yeah,sir.It's BJCXTH from China. 
Company's  chinese name is 北京创新通恒科技有限公司. 

 
 &gt;
 &gt; @eselmeister <https://github.com/eselmeister&gt;
 &gt; Our instrument is mainly used for separating and purifying 
 &gt; impurities.The instrument is a new product in our company.So we want 
 &gt; to find a software to control instrument and analyze the impurity signals.
 &gt;
 &gt; —
 &gt; <https://github.com/OpenChrom/openchrom/issues/24?email_source=notifications&amp;email_token=AAFHUT4Y5YSEYTB2ZS5MIVLQW5X7BA5CNFSM4JTGVR4KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEF4INGY#issuecomment-561546907&gt;, 
 &gt;
 
 ~~~~~~~~~~~~~~~~~~~~~~~~

eselmeister, 2019-12-09:
> Yeah,sir.It's BJCXTH from China.
> Company's  chinese name is 北京创新通恒科技有限公司.

Does the company has a website or similar for the product?

IMAUZS, 2019-12-09:
The website is www.bjcxth.com.
BJCXTH already has many products that separating and purifying impurities. 
And self-produced pumps and detectors and SMB system. 

 
 &gt; Yeah,sir.It's BJCXTH from China.&amp;nbsp;
 &gt; Company's&amp;nbsp; chinese name is 北京创新通恒科技有限公司.&amp;nbsp;
 &gt; ---Original---
 &gt; From: "eselmeister"<notifications@github.com&amp;gt;
 &gt; To: "OpenChrom/openchrom"<openchrom@noreply.github.com&amp;gt;;
 &gt; Cc: 
 &gt; "Mention"<mention@noreply.github.com&amp;gt;;"IMAUZS"<351092161@qq.com&amp;gt;;
 &gt; languages (#24)
 &gt; Is it a GC/MS or GC/FID from a known vendor like Shimadzu, Agilent?
 &gt; &amp;gt;
 &gt; &amp;gt; @eselmeister <https://github.com/eselmeister&amp;gt;
 &gt; &amp;gt; Our instrument is mainly used for separating and purifying
 &gt; &amp;gt; impurities.The instrument is a new product in our company.So we want
 &gt; &amp;gt; to find a software to control instrument and analyze the impurity 
 &gt; signals.
 &gt; &amp;gt;
 &gt; &amp;gt; —
 &gt; &amp;gt; 
 &gt; &amp;gt; 
 &gt; &amp;gt;
 &gt; spectrometry
 &gt; Dr. Philip Wenig » Founder » philip.wenig@openchrom.net »

Ramoonus, 2025-06-18:
sign me up for Dutch (NL)