OpenChrom/openchrom #25: Upgrade to new EJML version

laeubi, 2020-01-08:
A new EJML version with package version 0.38.x is available we should check if we can upgrade the NMR module to use this version as it includes some bugfixes and improvements.

@alexstark4 please have a look at this @lorenzgerber might can help with question regarding changed class/package names.

alexstark4, 2020-01-08:
NMR? Never heard of her! :)

alexstark4, 2020-01-13:
AFAICS, nothing has changed in the parts of the EJML used by the NMR features and everything works without errors.

laeubi, 2020-01-13:
What do you mean by "nothing changed"? At least the package imports need to be adjusted.

alexstark4, 2020-01-13:
Maybe I didn't express myself very clearly.

I adjusted the imports org.ejml.data and org.ejml.simple;version from "[0.29.0,0.30.0)"
to "[0.29.0,0.38.0)".

As a result I got no errors in the workspace and the features are working
as they did before. Is this procedure incorrect?

laeubi, 2020-01-13:
That way you exclude the 38 version leaving everything as is. Minimum version should be 0.38 and maximum should be 39 here `[0.38.0,0.39.0)`

alexstark4, 2020-01-14:
OK, I messed up with the brackets. I'll check it again.

commit 63e0d9df (2020-01-14): Fix #25
closed 2020-01-14 by alexstark4