OpenChrom/openchrom #40: MoleculeView: wrong structures

danield5732, 2020-04-20:
There still seems to be some errors in plotting the molecule structures. 

This is chemically unlikely:
[image]

This is wrong:
[image]

Maybe this is liked to #9

eselmeister, 2020-04-21:
The molecule structures are created using CDK and OPSIN. I'll investigate this.

eselmeister, 2020-04-22:
An option to type modify the settings and to type in IUPAC names, CAS# and/or SMILES have been added.

[image]
[image]
[image]
[image]

Iminostilbene
256-96-2
C1=CC=C2C(=C1)C=CC3=CC=CC=C3N2
https://pubchem.ncbi.nlm.nih.gov/compound/Iminostilbene#section=InChI-Key

2,2'-Dimethylbiphenyl
605-39-0
CC1=CC=CC=C1C2=CC=CC=C2C
https://pubchem.ncbi.nlm.nih.gov/compound/2_2_-Dimethylbiphenyl#section=2D-Structure

2,2'-Dimethylbiphenyl is still rotated.

closed 2020-04-22 by eselmeister

danield5732, 2020-05-18:
Thanks for the improvements and new options!

Just testing: Iminostilbene still has some issues

- SMILES is not calculated for Iminostilbene by Chromatogram Identifier - SMILES Identifier
- Therefore, Iminostilbene still show up with the wrong structure
- If SMILES code is inserted manually it works
- Changing the name to _5H-Dibenz[b,f]azepine_ or _2,2'-Iminostilbene_ leads to the right structure

Seems to be a tricky molecule or name :-)

Mailaender, 2021-03-22:
https://github.com/OpenChrom/openchrom/pull/105 adds support for Angstrom precise chemical structure data, so this will improve in the near future.

commit 92e0244e (2020-04-22): Fixed #40 - MoleculeView: wrong structures