OpenChrom/openchrom #42: Peak/Scan List: sorting does not work for Width and Name

danield5732, 2020-04-20:
Sorting the Peak/Scan List by clicking on the column header does not work propertly for the column Width and Name. 
The latter only works as aspected if all names are filled.

danield5732, 2020-04-20:
Feature request: It would be nice if the sorting could keep the order from before. 
Therefore, it would be possible to do combined sortings, e.g. first sort by S/N, second sort by classifier.
This would lead to the list sorted by classifier and within the classifier beeing stil sorted by S/N.

eselmeister, 2020-04-22:
Sorting by name or width has been fixed. There is currently no standard way for JFace tables to indicate multiple column sorting, see:

https://stackoverflow.com/questions/36542057/sort-jface-tableviewer-by-multiple-columns
https://bugs.eclipse.org/bugs/show_bug.cgi?id=133154

closed 2020-04-22 by eselmeister