OpenChrom/openchrom #48: Chromatogram Export "Template Chromatogram Report" loses peak entries

danield5732, 2020-04-20:
It is a very nice feature to combine peaks with the same name or CAS.

But this does not seems to be for peaks with close retention times only.
It also combines peaks with the same name, retention time difference of more than 4 minutes and with other peaks in between.
This leads to loosing peaks, since all peaks with sooner retention time are not present in the exported list.

danield5732, 2020-04-20:
There seems to be more issues.
Also adjacent peaks (Stop RT = Start RT) lead to recognition as one peak even with different names!

[image]

Chromatogram Report counts the peak "trans-(2-Chlorovinyl)dimethylethoxysilane" twice including the area of the first peak of "Cyclotrisiloxane, hexamethyl-". 
[image]
Maybe this is due to overlapping retention time windows in the exported Template. I attach this file, too.
[dd_190410_05b.txt](https://github.com/OpenChrom/openchrom/files/4502134/dd_190410_05b.txt)

I really appreciate this chromatogram report that compares reference chromatograms!
I am looking forward to the next update, coming soon.

Mailaender, 2021-03-23:
I used the DEMO chromatogram included in OpenChrom and found start = stop RT however they get reported without problem:

```
45.143 | 45.233 | Linolenic acid, 2-hydroxy-1-(hydroxymethyl)ethyl  ester (Z,Z,Z)- | 55268-58-1 | ALL
45.233 | 45.388 | Phthalic acid, hexyl oct-3-yl ester | 0-00-0 | ALL
```

Closing as this might already be fixed. If this is still a problem in the latest version, please reopen.

closed 2021-03-23 by Mailaender