OpenChrom/openchrom #55: Template PeakIdentifier: merging of peaks with the same name

danield5732, 2020-05-06:
Creating the template's .txt by "save as", the Template PeakIdentifier merges adjacent peaks with the same name. This I do not expect and I can not find any settings on this in the settings. 
Why is not every peak from peak/scan list exported? 
On which parameters the merge is based and can they be modified by the user, or turned off?

danield5732, 2020-05-06:
Maybe a correction: The peaks are not merged but lines are lost, if the actual name already exists.
This may be linked to #48

Mailaender, 2021-03-23:
My test case is as follows: open the demo chromatogram. Replace targets with unknown markers (no m/z) for adjacent peaks

[image]

The report contains the expected number of peaks included the unknowns:

```
40.366 | 40.560 | Adipic acid, isohexyl pent-4-en-2-yl ester | 0-00-0 |  |  |  | 129,111 | 
41.644 | 41.928 | Adipic acid, decyl pent-4-en-2-yl ester | 0-00-0 |  |  |  | 129,215 | 
43.219 | 43.361 | Unknown [] |  |  |  |  | 105,77 | 
44.601 | 44.756 | Unknown [] |  |  |  |  | 149,105 | 
45.143 | 45.233 | Unknown [] |  |  |  |  | 108,109 | 
45.233 | 45.388 | Unknown [] |  |  |  |  | 149,233 | 
```

This may already be fixed in the latest version or could be a specific problem that only appears with your data.

closed 2021-03-23 by Mailaender