OpenChrom/openchrom #58: Fix GSTL branch to support chemclipse integration builds

lgg4373, 2020-06-03:
The GSTL-1.1.x  branch (which I assume is the release branch?) does not comply with [https://github.com/eclipse/chemclipse/issues/344](https://github.com/eclipse/chemclipse/issues/344) and thus fails to build.

eselmeister, 2020-06-03:
We are working on the branch "develop" to integrate all improvements. So, if you'd like to compile a recent version, please checkout and build the develop branch.

closed 2020-06-03 by eselmeister

lgg4373, 2020-06-04:
Does this mean the sources for the current release of openchrom 1.4.x are available from the develop branch?

eselmeister, 2020-06-04:
Yes

eselmeister, 2020-06-04:
Have a look here how to setup the IDE:
https://github.com/OpenChrom/openchrom/wiki/Quickstart

Mailaender, 2020-11-25:
He was trying to package a release https://aur.archlinux.org/packages/openchrom/ to give some context here.

eselmeister, 2020-11-26:
I like the efforts to bundle OpenChrom for various OS.

I strongly recommend to only focus on the "develop" branch. This branch contains the latest activities.

Mailaender, 2022-07-16:
While compiling develop might get you the latest, important bits like most file converters will be missing. I updated https://aur.archlinux.org/cgit/aur.git/commit/PKGBUILD?h=openchrom&id=e7a1313294f6515536682962dec8d1245f361573 with some remarks on the quirks of how OpenChrom is released.