OpenChrom/openchrom #60: A handful of little things

danield5732, 2020-06-04:
There are some small issues I'm confrontied with regularely.
I'm giving them in a bunch, hope that is ok.

1. **Updating the views** (tabs or windows) is annoying. Even when a tab is still activated (Task Quick Access), but covered by another, it holds the content from its last use. This sometimes leads to confusions and is a big obstacle for productive work flows. Every active tab should show the current data selected, it should be updated at least if it is brought to the screen after beeing covered by another tab.
2. Overview Mode provides a tab named **Miscellanious Info** that can be filled with text. Anyway, the text is not saved at all. I wanted to use this for documentation for what was done with the data. Since the tab **Edit History** is not accessable by the user and it is not filled automatically by applying an filter to the data, which I would prefere.
3. **Saving an OCB file** performes quite slowly. While loading takes 5 s, saving needs about 80 s and breaks entusiastic work experiences ;-)
4. The **Targets** tab is updated each time an identification (NIST) is performed, but the new results are added. I would say it should be refreshed by default. Because the old targets may not be correct if the peak was modified before, e.g. by scan subtraction or modification of single mass traces.
5. **New legend in the Chromatogram Overlay** tab is very usefull but does not support the SIC very well. The m/z values are given just at the end of the Description column and due to long description they are not visible. I would love to have them visible more directly, e.g. in a separate column. Additionally, it would be nice if the description could be modified via settings/preferences since it is not saved. Therefore, also the Data Name rather than the file name may be selected.

I think these issues would enhance the user experience immediately. So I hope you can find some time to work on some of these. Thanks a lot!

Mailaender, 2021-03-22:
> I'm giving them in a bunch, hope that is ok.

One issue per issue please otherwise this becomes unmanageable.

2. Saving a chromatogram with misc header data as .ocb closes and reopen and the data is persisted:
[image]
Nothing is broken?

Mailaender, 2021-03-22:
3. Lower the compression in Window Preferences - Chromatography/Spectrometry > Converter > OpenChrom Converter > Measurement Data (*.ocb)

closed 2021-03-24 by eselmeister

Mailaender, 2021-03-24:
4. Is indeed a bug that was fixed by https://github.com/eclipse/chemclipse/pull/538.

5. Was improved a lot recently: https://github.com/eclipse/chemclipse/issues/542

[image]

[image]