OpenChrom/openchrom #69: Can't import ClarityChrom *.prm files

AMaywurm, 2020-06-23:
Hi everybody,

curious to try out OpenChrom since it looks very, I fail to import *.prm chromatograms generated with ClarityChrom 8. After trying to select chromatogram files or folder and clicking Finish, the progam promts "Please select at least one chromatogram".
What is also unexpected is that I am first asked to select files to import and then to select an import folder.

eselmeister, 2020-06-23:
Hi,

thanks for reporting this issue. Please contact us here and send us some 
test files.
https://lablicate.com/about/contact

Best,
Philip

elainfl, 2020-10-08:
Hello I have the same problem and I get the same message.

eselmeister, 2020-10-09:
@elainfl Thanks for your report. We will have a look at it, as soon as getting some files for testing.

elainfl, 2020-10-09:
Hello, Here I send you a few files.

Thanks.

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Remitente
notificado con
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
09/10/20
13:36:58

eselmeister, 2020-10-09:
@elainfl I assume the link has been removed by GitHub. Please use my email for sending the link again. I have sent you a reply this morning.