OpenChrom/openchrom #72: No peak identifier tool following windows download?

ShanePeterFitzgerald, 2020-07-09:
Hello! I downloaded openchrom 1.4 onto my windows laptop but I have to no option for the peak identifier? I have looked at settings and available plug ins but couldn't find any answers. I previously used openchrom on Mac and had no issues until the catalina update stopped me from using NIST! Would really appreciate some help! Cheers!
[image]

eselmeister, 2020-07-09:
> openchromissue

Please contact us here:
https://lablicate.com/about/contact

Lablicate is an official NIST distributor. Probably, we can help you 
getting your system up and running again.

Best,
Philip

ShanePeterFitzgerald, 2020-07-09:
Hi Philip

Thanks for the reply. I have sent my issue in thank you!

All the best
Shane

-- 
__

Séanadh Ríomhphoist/_

Email Disclaimer__
**

Tá an ríomhphost seo agus 
aon chomhad a sheoltar leis faoi rún agus is lena úsáid ag an seolaí agus 
sin amháin é. Is féidir tuilleadh a léamh anseo. 
<https://www4.dcu.ie/iss/seanadh-riomhphoist.shtml>  
<https://www4.dcu.ie/iss/seanadh-riomhphoist.shtml>*
_

This e-mail and any 
files transmitted with it are confidential and are intended solely for use 
by the addressee. Read more here. 
<https://www4.dcu.ie/iss/email-disclaimer.shtml> _
*_

closed 2020-07-30 by eselmeister