OpenChrom/openchrom #76: Manual or better help? https://github.com/OpenChrom/openchromcomp/wiki/OpenChrom-User-Manual

dukechem, 2020-08-04:
Thanks for all your work, but in trying to convince other chemists to try this, the Help/Documentation has problems below. 

The Help: Documentation launches a help-system with 3 top-level topics, and the first 2 are confusing chemists. Can we hide or move the first two topics?
  * Workbench User Guide
  * Java developement user guide 
  * Chromatography/Spectrometry <- only useful one (I think).

The first line (below) has broken link in overview for "Chromatography/Spectrometry"
```For help have a look at the wiki: https://wiki.openchrom.net. ```

Perhaps change the line to something like 
```"Updated help will be available soon, but until it is available, here is a link to a 2019 User manual:" ```
https://github.com/OpenChrom/openchromcomp/wiki/OpenChrom-User-Manual

That openchormecomp project has been archived, and the wiki-manual there not updated since 2019, but it is better than nothing.  

A quick-start guide would be useful... anyone know of any howto's for new users of openchrom?

eselmeister, 2020-08-06:
Thanks for your feedback.
The documentation is a bottleneck, indeed.

The initial plan of OpenChrom was to start the project and enable a 
community stepwise that takes care of writing and maintaining the help. 
Unfortunately, the community isn't big enough yet. Probably it will not 
be big enough in future as there are only a few people doing GC-MC, 
GC-FID ... in comparison to LibreOffice, Ubuntu users. Some docu is 
available here:

https://spectrometrylab.wordpress.com
https://github.com/openchrom/openchrom/wiki/OpenChrom-User-Manual

The videos on YouTube are outdated. We try to focus on the GitHub wiki. 
Additional help is available if one books a customer-specific workshop. 
We are open source enthusiasts, that's why we have founded and run 
OpenChrom. But we also have to earn our money. So far, we try to focus 
on making OpenChrom as intuitive as possible, with the resources we 
have. The flipside of the coin is, that the documentation suffers. I've 
not found a solution to solve this contradiction yet.

Best,
Philip

Mailaender, 2021-05-27:
I cleaned up the outdated wiki chapters, removed the unrelated help chapters in the application and added a hotkey cheat sheet. For a complete PDF user manual please https://lablicate.com/about/contact us.

closed 2021-05-27 by Mailaender