OpenChrom/openchrom #77: Peak/Scan List - sorting issue for Name and Classifier columns

danield5732, 2020-08-05:
There seems to be a sorting issue for the Peak/Scan List.
While Numbers are sorted correctly the Classifier column ist sorted weird.
[image]

For the Name column sorting is case-sensitive but it shouldn't. This is less intuitive since it leads to: 1...9 A...Z a....z
[image]

eselmeister, 2020-08-06:
@danield5732 Thanks for reporting. I'll have a look at it.

eselmeister, 2020-09-11:
@danield5732 Thanks for your report. This is fixed in the next version.
https://github.com/eclipse/chemclipse/issues/415

[image]

[image]

closed 2020-09-11 by eselmeister