OpenChrom/openchrom #82: Importing .wiff files produces really weird chromatograms

SaplingToss, 2020-09-09:
When I imported a .wiff file, the produced chromatogram was really strange. There were three issues. 

Firstly, the more minor issue is that the chromatogram does not scale and I end up with something that looks like this. 
[image]

Secondly, I think there is some corruption of data as some of the data more or less disappeared, seen in the discontinuous graph below.
[image]

Finally, additional peaks appeared that are not present when viewed using PeakView or SeeMs.

Hope this can be fixed!

eselmeister, 2020-09-09:
> image

Could you send me some test file?

Please contact me via:
https://lablicate.com/about/contact

Best,
Philip

SaplingToss, 2020-09-09:
Hi Phillip,

I have attached a .wiff file.

Ryan

eselmeister, 2020-09-09:
@SaplingToss Does the *.wiff file has been sorted out by GitHub?

SaplingToss, 2020-09-09:
@eselmeister Yes I think it was blocked. The .wiff file comes in 2 parts, I will zip them and upload.
[test.zip](https://github.com/OpenChrom/openchrom/files/5196700/test.zip)

eselmeister, 2020-09-09:
Alternatively, you could also provide it via DropBox or similar.

SaplingToss, 2020-09-10:
I uploaded it in my previous comment, can you see the zip file?