OpenChrom/openchrom #84: Can not add peak to ESTD

Seiken-2ch, 2020-09-25:
Hi. I followed the document [here](https://github.com/OpenChrom/openchrom/wiki/Quantitation-ESTD) to export peaks to database. Everything goes well until the "Open the database, activate the external standards parts and execute the response factor calculation" part. There are nothing added and database is empty. Also I have checked the .ocq file in Windows file manager, it's updated without size increase when I added the peak.

eselmeister, 2020-09-26:
Hi Seiken,

have you integrated the peak area, e.g. via the "Trapezoid Integrator" 
before adding the peak to the quantitation database?

Best,
Philip

Seiken-2ch, 2020-09-28:
Yes I Integrated my peaks. Also my files are CSDs from a HPLC and your example was made with MSD. I don't know if this matters.

eselmeister, 2020-09-28:
That's the reason. The external quantitation is focused on MSD data at 
the moment. Principally, CSD could be supported too, but this requires 
some efforts to adjust the code. Generally, there should be a notice, 
that CSD and WSD is not supported yet, when adding peaks to the 
quantitation database.

Seiken-2ch, 2020-10-02:
Okay I see. Anyway I added the peak name, unit, RT, and concentration-response pair values manually with Cross Zero and got a response/concentration trend line, but some quantification with this ESTD returns negative value, with a positive response. Is this an unexpected behavior or I made some mistake? Or still this could not be expected works normally?
[image]
　　⇑
Manually constructed ESTD
Quantification result
　　⇓
[image]

eselmeister, 2020-10-02:
@Seiken-2ch Could you provide the *.ocq ESTD database and the measurement. I will inspect it. If files shouldn't be shared publicly, you could send me the files also directly. In such a case, please contact me first via the contact form:

https://lablicate.com/about/contact

Seiken-2ch, 2020-10-05:
Here is the relative files.
[issue_files.zip](https://github.com/OpenChrom/openchrom/files/5328970/issue_files.zip)

Also, is it possible to view the function of calibration curves?

Mailaender, 2021-11-29:
There is a commercial addon which simplifies things and it also shows calibration functions. https://lablicate.com/about/contact us if you are interested.

closed 2022-12-13 by Mailaender