OpenChrom/openchrom #85: Some {extension} tag do not work properly

Seiken-2ch, 2020-09-25:
The {extension} tag at some file export menu returns a string without a leading dot, while others with. Which make it generate files without extension. For example, "Template Chromatogram Report (*.txt)" has this problem. Also I believe that I encountered same problem some else where.

[image]
⇓
[image]

eselmeister, 2020-09-26:
Yep, thanks for reporting. We had this issue with other template methods too:
https://github.com/OpenChrom/openchrom/issues/47

I'll check this.

eselmeister, 2020-09-26:
This is fixed in the latest version:
https://lablicate.com/platform/openchrom/download

[image]

closed 2020-09-26 by eselmeister