OpenChrom/openchrom #88: Feature find or request: report multiple datas at one table

Seiken-2ch, 2020-10-03:
It's good to have a function to report datas in multiple files as a single table to paste into spreadsheet programs, and I did not find it yet.

I found that "Template Chromatogram Report" with "Append" option could report datas within a file,  but it seems there are no option to remove the header and the tags, which made it could not become a table. Maybe a "report in single line" or "data only" option works directly. If there are still no function like this.

eselmeister, 2020-10-03:
A chromatogram could contain reference(s). They are either loaded directly via the converter (Agilent ->MSD & FID) or added manually. Chromatograms in open editors can be added:

[image]

The peak template report should match your needs. It can be created via the preferences editor. The editor allows an option to export the data to text format.

[image]

When doing a report, run it from the context menu. First try it with all column. Afterwards, you can specify which columns to print:

[image]

Seiken-2ch, 2020-10-06:
Thank you. Is it possible to report quantification results in this way? and could I disable the tags of columns and the extra line break, only get a column of numbers?

[image]