OpenChrom/openchrom #93: MicrobeMS .pkf Tests

eselmeister, 2020-10-19:
The MicrobeMS fail. When running it locally, the following error log is generated:

```
java.lang.NoSuchMethodError: java.nio.ByteBuffer.limit(I)Ljava/nio/ByteBuffer;
	at us.hebi.matlab.mat.types.Sources$ByteBufferSource.readBytesAsStream(Sources.java:280)
	at us.hebi.matlab.mat.types.AbstractSource.readInflated(AbstractSource.java:177)
	at us.hebi.matlab.mat.format.Mat5Reader.readEntry(Mat5Reader.java:270)
	at us.hebi.matlab.mat.format.Mat5Reader.readMatContent(Mat5Reader.java:228)
	at us.hebi.matlab.mat.format.Mat5Reader.readMat(Mat5Reader.java:194)
	at us.hebi.matlab.mat.format.Mat5.readFromFile(Mat5.java:53)
	at net.openchrom.msd.converter.supplier.pkf.converter.io.PkfReader.read(PkfReader.java:47)
	at net.openchrom.msd.converter.supplier.pkf.converter.DatabaseImportConverter.convert(DatabaseImportConverter.java:40)
	at net.openchrom.msd.converter.supplier.pkf.converter.MassSpectrumImportConverter_Ecoli_ITest.setUp(MassSpectrumImportConverter_Ecoli_ITest.java:36)
	at junit.framework.TestCase.runBare(TestCase.java:139)
	at junit.framework.TestResult$1.protect(TestResult.java:122)
	at junit.framework.TestResult.runProtected(TestResult.java:142)
	at junit.framework.TestResult.run(TestResult.java:125)
	at junit.framework.TestCase.run(TestCase.java:129)
	at junit.framework.TestSuite.runTest(TestSuite.java:252)
	at junit.framework.TestSuite.run(TestSuite.java:247)
	at org.junit.internal.runners.JUnit38ClassRunner.run(JUnit38ClassRunner.java:86)
	at org.eclipse.jdt.internal.junit4.runner.JUnit4TestReference.run(JUnit4TestReference.java:86)
	at org.eclipse.jdt.internal.junit.runner.TestExecution.run(TestExecution.java:38)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.runTests(RemoteTestRunner.java:538)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.runTests(RemoteTestRunner.java:760)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.run(RemoteTestRunner.java:460)
	at org.eclipse.pde.internal.junit.runtime.RemotePluginTestRunner.main(RemotePluginTestRunner.java:229)
	at org.eclipse.pde.internal.junit.runtime.PlatformUITestHarness.lambda$0(PlatformUITestHarness.java:43)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:37)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:182)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4915)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4521)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1170)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1059)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:667)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:597)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.pde.internal.junit.runtime.NonUIThreadTestApplication.runApp(NonUIThreadTestApplication.java:52)
	at org.eclipse.pde.internal.junit.runtime.UITestApplication.runApp(UITestApplication.java:43)
	at org.eclipse.pde.internal.junit.runtime.NonUIThreadTestApplication.start(NonUIThreadTestApplication.java:46)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:656)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:592)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1498)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1471
```

The CI also produces errors:

`Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 0.092 s <<< FAILURE! - in net.openchrom.msd.converter.supplier.pkf.converter.MassSpectrumImportConverter_Ecoli_ITest`

eselmeister, 2020-10-20:
See:
https://github.com/OpenChrom/openchrom/pull/92

Mailaender, 2020-10-21:
Sorry for the hassle, but I can't reproduce the problem locally:

[image]

Mailaender, 2020-10-21:
```
eclipse.buildId=unknown
java.version=11.0.8
java.vendor=N/A
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=de_DE
Framework arguments:  -version 3 -port 33075 -testLoaderClass org.eclipse.jdt.internal.junit4.runner.JUnit4TestLoader -loaderpluginname org.eclipse.jdt.junit4.runtime -classNames net.openchrom.msd.converter.supplier.pkf.AllTests -application org.eclipse.pde.junit.runtime.coretestapplication -testpluginname net.openchrom.msd.converter.supplier.pkf
Command-line arguments:  -os linux -ws gtk -arch x86_64 -consoleLog -version 3 -port 33075 -testLoaderClass org.eclipse.jdt.internal.junit4.runner.JUnit4TestLoader -loaderpluginname org.eclipse.jdt.junit4.runtime -classNames net.openchrom.msd.converter.supplier.pkf.AllTests -application org.eclipse.pde.junit.runtime.coretestapplication -data /home/matthias/Entwicklung/Eclipse/openchrom-workspace/../junit-workspace -dev file:/home/matthias/Entwicklung/Eclipse/openchrom-workspace/.metadata/.plugins/org.eclipse.pde.core/pde-junit/dev.properties -os linux -ws gtk -arch x86_64 -consoleLog -testpluginname net.openchrom.msd.converter.supplier.pkf
```

I use Java 11 but https://github.com/HebiRobotics/MFL requires Java version 1.6 (only the tests require version 7 and 8) so that does not seem to be the problem.

laeubi, 2020-10-22:
This can happen if [MFL](https://github.com/HebiRobotics/MFL) is compiled with a more recent JDK (e.g. Java 9+) even if the class-file-compatibility might be 1.6 ... 

See for example https://github.com/eclipse/jetty.project/issues/3244 and stuart-marks comment [for a solution](https://github.com/eclipse/jetty.project/issues/3244#issuecomment-495322586).

Mailaender, 2020-10-22:
Ah, yes the MFL was self compiled by me using OpenJDK 11.

eselmeister, 2020-10-22:
@Matthias Is it possible for you to compile the library using Java 8?

Mailaender, 2020-10-22:
Yes, @openSUSE has a wide variety of Java packages including 1.8.0 which I now downgraded to.

eselmeister, 2020-10-23:
Perfect. I'll check the PR. I would like to upgrade to the latest 
Eclipse release and the latest Java version better sooner than later. 
Currently, JavaFX is used, which needs to be removed first. There is 
also an issue with GTK3 and SWT, which also needs to be solved first. So 
I assume, an upgrade can be scheduled for Q2/2021.

closed 2020-10-23 by eselmeister