OpenChrom/openchrom #98: Manual peak detection

Ejtnaj, 2021-02-12:
Dear OpenChrom-Team,

I started using OpenChrom just recently, and I really like it so far, thank you for all your work! I have a question about peak detection though:
I've been trying to detect peaks manually, but unfortunately I cannot find out how. When I click "Show Peak Detector" a new perspective opens, but my chromatogram doesn't show up in there. Also when I go to Preferences > A) Chromatography/Spectrometry > Peak Detector > Manual Peak Detector, there is no info or options I can select. Do I need to install an additional plugin? I'm using version 1.4.x for Linux as well as Windows, my files are ChemStation files (.ch), and I use a DAD. I would be very happy if you could help me and tell me what I'm doing wrong.
Best regards,

Ejtnaj

eselmeister, 2021-02-12:
@Ejtnaj Please provide some test files via Lablicate. First contact us via the contact form. Then I'll create a share folder.
https://lablicate.com/about/contact

closed 2021-02-24 by eselmeister

eselmeister, 2021-02-24:
@Ejtnaj I have added a template/manual peak detector for UV/Vis and DAD. Currently, it uses the TIC. An option to specify traces exists, but this is not enabled yet. Functionality will follow.

[image]

[image]

[image]

[image]

A new version will be available in ~2 hours.

commit 8e6257e2 (2021-02-24): Fixed #98 - Manual peak detection