OpenChrom/openchrom #99: More specific version download?

teknowledgist, 2021-02-14:
Hi.  I'm an OSS advocate and my work in academia has me supporting a lot of software used in Chemistry (and other) research.  *I am not an OpenChrom user,* but I'm trying to maintain a [Chocolatey package for OpenChrom](https://chocolatey.org/packages/openchrom) to make it easier for users who want to try OpenChrom or for lazy sysadmins like myself who are tasked with installing it on every machine in a lab.  If you aren't aware, [Chocolatey](https://chocolatey.org) is like `apt` or `RPM` but for Windows.

I'm having trouble identifying (via script) the version that folks would be getting when they request Chocolatey to get/install OpenChrom.  All the website has is "1.4.x", and this GitHub site doesn't identify the build or release either.  Do you think you could include posting a version ID/log into your build process so everyone knows when a download will be a new version?

Thanks!

eselmeister, 2021-02-14:
Please contact us here:
https://lablicate.com/about/contact

Mailaender, 2021-11-29:
https://github.com/OpenChrom/openchrom/wiki/changelog contains version plus time stamps if that helps.

teknowledgist, 2021-11-29:
@Mailaender Thanks for pointing that out.  

Unfortunately though, it is not clear that those timestamps have anything to do with [the download ](https://www.lablicate.com/platform/openchrom/download).  

As it is, I can see that the checksum changes, but the version isn't listed.  If those timestamps (with change documentation) are being auto-loaded to GitHub, why can't the zipped binaries be loaded as releases too?

Mailaender, 2021-11-29:
The changelog is written manually the releases are not based on tags. A new release overwrites the old one.

teknowledgist, 2021-11-30:
OK.  That doesn't really resolve the issue though because there is no definitive connection between what someone manually adds to the wiki on one site and what is available to download on a completely different site.

How can I know that what can be downloaded [from here](https://www.lablicate.com/platform/openchrom/download) at any moment is the version entered [at the top of this page](https://github.com/OpenChrom/openchrom/wiki/changelog)?  

Am I making any sense?

Mailaender, 2021-12-06:
I understand your frustration. We would have to revamp our deployment. Alternatively you could give me permission https://community.chocolatey.org/profiles/Mailaender so we can set hash/date from within our private build servers.

teknowledgist, 2021-12-06:
As the application author(s), of course you can be a maintainer of the package.  If you want to build/submit the packages as part of your build process, feel free.  Let me know if I can help.

Mailaender, 2021-12-11:
You only need to add me as maintainer.

teknowledgist, 2021-12-12:
I have already submitted your name for maintainership of the package.  You should have gotten an email from chocolatey.org with a link you have to follow to accept it.  Let me know if you didn't get it and I'll try to redo it.

Mailaender, 2021-12-12:
I didn't get it.

Mailaender, 2021-12-16:
I figured this out with @chocolatey staff. I disabled e-Mail notifications or anti-spam measures did not pass the confirmation e-Mail through.

Anyway, https://community.chocolatey.org/packages/openchrom/1.4.0.20211210 was now submitted by the semi automated https://github.com/OpenChrom/OpenChromChocolateyPackage/actions/runs/1588477389 so this should be good to go now. You have access to the repository so you can still edit the packaging or metadata if you like. The `.nuspec` is based on yours only with some minor updates. Throwing in new versions is now my job. I hope this helps cope with our aggressive rolling-release style.

closed 2021-12-16 by Mailaender

teknowledgist, 2021-12-16:
Super!  Thank you for working out a solution.