OpenChrom/openchrom #101: Data Explorer - User Location is bucking strongly

danield5732, 2021-02-25:
Scrolling through the User Location (We have in their about 800 folders) ist strongly bucking.
This is new, maybe came with the new Task Bar.
In our previous version it scrolled smoothly.

Mailaender, 2021-03-19:
Is the user location on a network drive?

danield5732, 2021-03-19:
Yes it is!
Sorry for scarce information.

There was a similar issue two years ago, but untill the last update it worked quick and fine.

Mailaender, 2021-03-19:
Which timestamp did the last version have?

Mailaender, 2021-03-19:
What kind of file formats are in the folder?

danield5732, 2021-03-21:
I think timestamp of the previous version is 1.4.0.202009110856

Mostly Agilent raw data files/folders are in the user location.

Mailaender, 2021-04-28:
If you head into the file explorer settings and untick categories, will it get better?

danield5732, 2021-04-29:
Yes it is, absolutely! Deactivating the categories on CSD, WSD, NMR, etc. clearly speed up the File explorer with MSD data.
Thank you!

Mailaender, 2021-07-05:
If you download the latest OpenChrom release, it will start to measure the performance of our converter plugins as soon as you hit the data explorer. When it performs really bad you can send us your `ChemClipse.log` file so we can try to identify the perpetrators.

Mailaender, 2022-04-08:
Closing due to lack of activity.

closed 2022-04-08 by Mailaender

Mailaender, 2022-11-14:
https://github.com/eclipse/chemclipse/issues/1186 should improve the network drive situation on Windows.