OpenChrom/openchrom #102: Task Bar: user specific mandatory and additional entries?

danield5732, 2021-02-25:
It is a nice feature to differenciate manadatory and additional windows to display.
However, there come up two issues:

- If I want to display a window (e.g. Molecule) by default, this is not possible, because it is allocated to "additional" and I have to activate it by two addtional clicks.
- If I do not want to display a window by default that is allocated to "mandatory", I have to set it to "--". Consequently, it is no more accessable via the Taks Bar and I would have to activate it with at least 4 addional clicks.

Therefore, I would prefere to be able to define which window is mandatroy or addional.

eselmeister, 2021-03-26:
@danield5732 I understand your request. That's a good point. We will investigate how to improve the toolbar functionality.

Mailaender, 2021-07-05:
To ease your pain for the moment: the molecule view is now enabled by default. https://github.com/eclipse/chemclipse/issues/619