OpenChrom/openchrom #109: Shifting Y-axis in a chromatogram overlay changes the x-axis units (Mac OS, High Sierra)

parkashrawat, 2021-05-12:
[image]
[image]

Issue/bug: Shifting Y-axis in a chromatogram overlay changes the x-axis units. It also happens with individual plots while shifting the y-axis. Default settings are not changed.

eselmeister, 2021-05-12:
@parkashrawat Thanks for reporting. Is the x axis displayed correctly again if you select "Reset 1:1" in the chart via the pop-up menu?

parkashrawat, 2021-05-12:
> @parkashrawat Thanks for reporting. Is the x axis displayed correctly again if you select "Reset 1:1" in the chart via the pop-up menu?

Yes, after the 1:1 reset the axis shows correctly.

eselmeister, 2021-05-22:
@parkashrawat Thanks for your report. This is fixed and will be released with the next version.

closed 2021-05-22 by eselmeister