OpenChrom/openchrom #112: Cannot extract single ion chromatogram

NitsujiM, 2021-05-26:
The 2011 YouTube video "OpenChrom - Mass Spectrum Selection" shows how to select a mass from the spectrum to generate a "Selected nominal m/z Chromatogram". I cannot figure-out how to get this feature working or how to extract single ion chromatograms in general in the current version of OpenChrom. If this feature has been removed, is it possible to restore it in a future release? If the ability to extract ion chromatograms is indeed available somehow, please let me know. I speak for many fellow colleagues who have tested OpenChrom and like it, but do not continue to use it simply because of the absence of this feature. Thank you.

Mailaender, 2021-05-27:
The YouTube videos are very old and should probably be deleted as the user interface changed several times since they got uploaded. For the single ion chromatogram click "Data Analysis", in the "Data Explorer" open your files. In the toolbar click "Overlay"

[image]

In the bottom left a new window will appear where you can select "SIC" in the dropdown box. You can maximize and move this window freely. It is also possible to add your own ion traces.

[image]

I hope this is helpful. For a complete user manual please https://lablicate.com/about/contact us.

closed 2021-05-27 by Mailaender

NitsujiM, 2021-05-28:
This was extremely helpful! Thanks!