OpenChrom/openchrom #117: Cannot move past welcome screen Mac OS 11.4

tyrochemist, 2021-06-21:
Hello,

OpenChrom worked on my previous installation of Mac OS but has stopped after upgrading to Big Sur. It will launch successfully, but when I click on any of the panes to open a perspective from the welcome screen, the top bar changes but nothing else happens.

I ran the program from terminal and didn't see much even when clicking on the panes, until I quit the program, at which point several lines were returned. The main one seems to be:

`***WARNING: Display must be created on main thread due to Cocoa restrictions. Use vmarg -XstartOnFirstThread`

I also received the warning:
`<<< MARX-Dongle Support can't be activated, check system-configuration! (java.lang.UnsatisfiedLinkError: no jnicbios in java.library.path) >>>`

I am not sure if there is a fix to this that can be made, as I have lost all functionality of the program as it stands. 

Thanks for looking at this!

zacharyzollman, 2021-07-03:
Hi! I have had this issue as well. In November of last year, I filled out the Lablicate contact form and received a reply saying that fixing the problems caused by Big Sur would take "a while." They included [this link](https://www.eclipse.org/forums/index.php/t/1104820/), which I didn't find very helpful. I'm not very optimistic about a full restoration but I hope there will at least be a partial workaround soon.

eselmeister, 2021-07-03:
@zacharyzollman Yep, I understand that you are frustrated. We are struggling with this issue since a long while too.

eselmeister, 2021-07-07:
You could try to bypass the welcome screen and directly switch to the "Data Analysis" perspective. Close OpenChrom, modify the `openchrom.ini` and restart OpenChrom. Append the following line:

`-Dapplication.perspective=org.eclipse.chemclipse.ux.extension.xxd.ui.perspective.main`

zacharyzollman, 2021-07-08:
> You could try to bypass the welcome screen and directly switch to the "Data Analysis" perspective. Close OpenChrom, modify the `openchrom.ini` and restart OpenChrom. Append the following line:
> 
> `-Dapplication.perspective=org.eclipse.chemclipse.ux.extension.xxd.ui.perspective.main`

@eselmeister thanks so much for the suggestion! This allowed me to get past the welcome screen but I was unable to open any files or see any file locations (such as the User Location). I was also unable to open any of the nested items in preferences. The tab names were missing as well. Lmk if there are any other things you want me to check. I was running version 1.4.0.202107050846. I appreciate your help!

JMRoper, 2021-08-02:
Hi everyone! I was just checking up on this issue to see if any progress was made. I am running an old operating system as to not loose by ability to run the program so I can not check for myself if the issue has been resolved.

Mailaender, 2021-08-02:
We started some platform update and maintenance to resolve these issues, but I expect them to take a while.

Mailaender, 2022-03-27:
Good news: this is finally fixed. Head over to https://lablicate.com/platform/openchrom/download and download version 1.5.0!

closed 2022-03-27 by Mailaender

zacharyzollman, 2022-06-21:
Thank you for fixing this! I appreciate it.