OpenChrom/openchrom #130: CSV or OCB file doesn't open in OpenChrom

octavianparascinet, 2021-08-10:
Hi everyone! 
I have a problem here. I'm using OpenChrom for analyzing some chromatograms and identifying/quantifying their peaks. Everything goes normally with the manual detection peaks (I've used Peak Detector UI) and the integration, but when I save the chromatogram in csv or ocb format it doesn't open in the app. I don't know how to recover all the information (detection and integration of peaks) from the saved files.

eselmeister, 2021-08-10:
Which OpenChrom version do you use?
Could you supply a *.ocb file?

eselmeister, 2021-08-12:
@octavianparascinet Thanks for providing the demo file. I'm able to load it:

[image]

Could you send me a screenshot of your version?

octavianparascinet, 2021-08-16:
Ok, but how do I get back all the detected peaks with their RT and their percentage area? Because it does open to me as well, as you see in the screenshot I attached. But there's no information about these already measured parameters. 
[image]

octavianparascinet, 2021-08-30:
And how do I read the .ocb file in OpenChrom. Nothing appears on the interface when I do so. Sorry I'm inexperienced with this software and I need a little help. Thank you for your replies!

eselmeister, 2021-08-30:
Please do the following:

* open a csv file with your data
* run a peak detection / integration on your chromatogram
* save as... the chromatogram in *.ocb format
* close the chromatogram
* open the *.ocb chromatogram

Please create screenshots of all the above mentioned steps.

closed 2021-09-17 by Mailaender