OpenChrom/openchrom #166: Manual detection peak

octavianparascinet, 2021-11-29:
How can I manually detect a peak? I'm doing the manual detection on the peak detector window and when I press the "Add the manually detected peak" button it doesn't enter the list. I need to do anything else or why it doesn't work? I'm attaching a screenshot for you to see that the baseline is drawn and the peak doesn't appear on the "peak list" top left. 
[image]
.

Mailaender, 2021-11-29:
You can press the `Peaks` button in the toolbar and use the `Peak Detector` view, which is the old style way and may be deprecated at some point. The modern alternative is to right-click on to the chromatogram, select `Peak Detector` → `Peak Detector UI [Direct]` which shows existing peaks and also allows for manual peak detection. Both require you to hold the <kbd>CTRL</kbd> modifier.

closed 2021-11-29 by Mailaender

octavianparascinet, 2021-11-29:
Ok, thank you. It worked perfectly fine with the Detector UI (Direct), but not the old style. I have another issue about saving the detected peaks with the integration area. I don't know how to do this. Everytime I save the file in .ocb format and I open it in openchrom, it only appears the chromatogram, it doesn't show the already detected peaks nor the integration area.

Mailaender, 2021-11-29:
Please https://lablicate.com/about/contact so we can exchange some of the problematic data as `.ocb` should save all the peaks.