OpenChrom/openchrom #168: Peak alignment

octavianparascinet, 2021-11-29:
Hello! How can I do a peak alignment between up to 40 different samples? I saw there's an overlay view for the chromatograms, and there's also an alignment toolbar option. I press some of the alignment functions but it seems to me that it doesn't do nothing when I look back to the overlay perspective. If you could give me an example it'll work.

Mailaender, 2021-11-30:
Take a look at https://lablicate.com/marketplace/extensions/chromident but it does not support alignment by warping to the chromatogram, it only matches the peaks regardless of their retention time.

octavianparascinet, 2021-12-01:
Ok, I understand. Therefore every time I'll be doing the alignment of the chromatograms there will be no stretching nor shrinkage of the different peaks. The function will only be detecting same peaks with similar retention times from the different chromatograms. But after alignment and detection of peaks from each chromatogram, will the identification function let me recognize automatically the same peaks as the same compound?

Mailaender, 2021-12-01:
Yes, it matches the peaks from each chromatograms by mass spectra (even if the compound names differ slightly). See https://youtu.be/Hz6ti-Ed6LE

octavianparascinet, 2021-12-01:
That's the problem. We only have the chromatograms because we just did the HPLC method on our samples. In that case it cannot work?

Mailaender, 2021-12-01:
In that case, it is the wrong tool. Sorry.

Mailaender, 2025-09-14:
We added peak alignment to https://lablicate.com/marketplace/extensions/powertools which is a commercial add-on for OpenChrom.

closed 2025-09-14 by Mailaender