OpenChrom/openchrom #172: the Bruker file to not be recognized

TGQgood, 2021-12-02:
I would like to ask what caused the Bruker file to not be recognized. Area1 is a project generated by build, and area2 is a downloaded project. Thank you!
[image]

Mailaender, 2021-12-02:
Where did you download the files from?

TGQgood, 2021-12-02:
**### The area1 project is downloaded from gitHub, follow wiki build**
[image]
[image]

**### The area2 project is downloaded from this page**
[image]

**### I'm using the Windows operating system**

Mailaender, 2021-12-02:
I see. That is to be expected. See also my change https://github.com/OpenChrom/openchrom/commit/b2698dccfdbf5162c04d8de22a324b456ca0a90a which explains that the Lablicate provided edition contains the proprietary converters.

commit 80577c84 (2021-12-02): Clarify editions and the lack of converters.

Closes #172
closed 2021-12-02 by Mailaender