OpenChrom/openchrom #175: Malformed mzML output from .D file conversion

bspratt, 2021-12-07:
Greetings, fellow open source developers! A Skyline user recently presented an mzML file supposedly converted from Agilent .D with OpenChrom. It's an invalid file due to a mismatch in the declared and actual lengths of the encoded binary data arrays. This value needs to be correct so that parsers can efficiently allocate buffers when reading large scans,

Here's the start of that file:

&lt;?xml version="1.0" encoding="UTF-8" standalone="yes"?&gt;
&lt;mzML xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" version="1.1.0" xsi:schemaLocation="http://psi.hupo.org/ms/mzml http://psidev.info/files/ms/mzML/xsd/mzML1.1.0.xsd"&gt;
&lt;run startTimeStamp="2021-11-22T02:05:00.000-07:00"&gt;
&lt;spectrumList&gt;
&lt;spectrum defaultArrayLength="0"&gt;
&lt;cvParam accession="MS:1000511" value="1" name="ms level"/&gt;
&lt;cvParam accession="MS:1000127" name="centroid spectrum"/&gt;
&lt;scanList&gt;
&lt;scan&gt;
&lt;cvParam accession="MS:1000016" value="305545" name="scan start time" unitAccession="UO:0000028" unitName="millisecond"/&gt;
&lt;/scan&gt;
&lt;/scanList&gt;
&lt;binaryDataArrayList&gt;
&lt;binaryDataArray arrayLength="2" encodedLength="20"&gt;
&lt;cvParam accession="MS:1000574" name="zlib compression"/&gt;
&lt;cvParam accession="MS:1000523" name="64-bit float"/&gt;
&lt;cvParam accession="MS:1000514" name="m/z array"/&gt;
&lt;binary&gt;eJybNRMIKqMcZoHonUEOAEnyCFs=&lt;/binary&gt;
&lt;/binaryDataArray&gt;
etc

that "encodedLength" value needs to agree with the actual length of the string in the "binary" element.

I'd also note that there's a lot of missing information about the source of the original data, who/what/how the conversion was done, etc, but that's not the immediate issue.

I hope this helps!

Mailaender, 2021-12-07:
Can you share the original file? I am not sure which array and encoded length would be correct in this case.

bspratt, 2021-12-07:
I'll have to clear that with the user. 

But I don't think you really need it - the important thing is that the encoded representation has more characters than declared. This trips up parsers since they use that declaration to preallocate space to hold the base64 string for processing.

bspratt, 2021-12-07:
OK, here's the data: https://skyline.ms/announcements/home/support/download.view?entityId=d5d707bc-39ab-103a-9d0c-22f5355625be&name=TMS2047.D.zip

and the result of the conversion with OpenChrom:  https://skyline.ms/announcements/home/support/download.view?entityId=d5d707bc-39ab-103a-9d0c-22f5355625be&name=TMS2047.mzML

Mailaender, 2021-12-08:
I see. We currently write the actual number of bytes of compressed data written to the output buffer as `encodedLength`.

closed 2021-12-08 by eselmeister

bspratt, 2021-12-08:
Sounds good. Have you tried running the new result through a reference tool like msconvert or SeeMS from proteowizard? Thanks for jumping on this!

Mailaender, 2021-12-08:
`msconvert` segfaults for me on @openSUSE Leap 15.3

> msconvert: loadlocale.c:129: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.

bspratt, 2021-12-08:
Shoot me the mzML file and I'll have a look. Thanks!

Mailaender, 2021-12-08:
https://share.lablicate.com/index.php/s/FaLeGSsjBKSYNzF

bspratt, 2021-12-08:
Not there yet... (running on Windows, which seems to use libraries that are a bit more graceful handling errors)

>msconvert.exe -z --mzXML "TMS2047 (1).mzML"
format: mzXML
    m/z: Compression-Zlib, 64-bit
    intensity: Compression-Zlib, 32-bit
    rt: Compression-Zlib, 64-bit
ByteOrder_LittleEndian
 indexed="true"
outputPath: .
extension: .mzXML
contactFilename:
runIndexSet:

spectrum list filters:

chromatogram list filters:

filenames:
  TMS2047 (1).mzML

processing file: TMS2047 (1).mzML
calculating source file checksums

writing output file: .\TMS2047 (1).mzXML
[SpectrumWorkerThreads::work] error in thread: [IO::HandlerBinaryDataArray] At position 1124: expected array of size 2, but decoded array is actually size 1.
[SpectrumWorkerThreads::work] error in thread: [BinaryDataEncoder::copyBuffer()] Bad byteCount.
[SpectrumWorkerThreads::work] error in thread: [BinaryDataEncoder::copyBuffer()] Bad byteCount.
[SpectrumWorkerThreads::work] error in thread: [BinaryDataEncoder::copyBuffer()] Bad byteCount.
[SpectrumWorkerThreads::work] error in thread: [BinaryDataEncoder::copyBuffer()] Bad byteCount.
[SpectrumWorkerThreads::work] error in thread: [BinaryDataEncoder::copyBuffer()] Bad byteCount.
[SpectrumWorkerThreads::work] error in thread: [BinaryDataEncoder::copyBuffer()] Bad byteCount.
[SpectrumWorkerThreads::work] error in thread: [IO::HandlerBinaryDataArray] At position 7429: expected array of size 2, but decoded array is actually size 1.

Mailaender, 2021-12-08:
Is that a complaint about `EncodedLength` or `ArrayLength`?

bspratt, 2021-12-08:
At least one problem is here:
<cvParam accession="MS:1000523" name="32-bit float"/>
that accession number should be
MS:1000521

bspratt, 2021-12-08:
ugh, markdown ate the text:

cvParam accession="MS:1000523" name="32-bit float"

you're using the same cvparam for 64 bit and 32 bit

bspratt, 2021-12-08:
Tidy that up, and msconvert complains about this:

<binaryDataArray arrayLength="0" encodedLength="12">
<cvParam accession="MS:1000574" name="zlib compression"/>
<cvParam accession="MS:1000523" name="64-bit float"/>
<cvParam accession="MS:1000514" name="m/z array"/>
<binary>eJwDAAAAAAE=</binary>
</binaryDataArray>

it's not wrong, exactly, but it's unusual to go to all that trouble to encode an empty array.

Mailaender, 2021-12-09:
Okay, I got it so far that the MS Convert GUI does something with the file and SeeMS opens it (though it doesn't look quite right) https://share.lablicate.com/index.php/s/gH7r3JMmarXMxna

I initially only tested against http://code.google.com/p/jmzml/ which is very forgiving.

Mailaender, 2021-12-10:
I haven't managed to get https://build.opensuse.org/package/show/home:Mailaender:branches:science/openms to compile as it requires a lot of dependencies.

We released a new version https://github.com/OpenChrom/openchrom/wiki/changelog#202112100257 including these hotfixes.

bspratt, 2021-12-16:
You're getting close. The "index" attribute of the "spectrum" element is 0-based, so your file should start out 

&lt;?xml version="1.0" encoding="UTF-8" standalone="yes"?&gt;
&lt;mzML xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" version="1.1.0" xsi:schemaLocation="http://psi.hupo.org/ms/mzml http://psidev.info/files/ms/mzML/xsd/mzML1.1.0.xsd"&gt;
&lt;run id="chromatogram@1eb355c_1639055342512" startTimeStamp="2021-11-22T02:05:00.000+01:00"&gt;
&lt;spectrumList&gt;
&lt;spectrum id="scan=1" **index="0"** defaultArrayLength="0"&gt;
&lt;cvParam accession="MS:1000511" value="1" name="ms level"/&gt;

ProteoWizard still doesn't like those zero length binary data arrays, but I don't think they're strictly illegal so I'll have to make a code change to tolerate those.

bspratt, 2021-12-16:
Also, the array length must be encoded in defaultArrayLength rather that in the array declaration itself, unless the binary array actually does have a different length than declared in defaultArrayLength. (This is very rare, and never the case for common data types such as m/z, intensity etc). See https://raw.githubusercontent.com/HUPO-PSI/mzML/master/schema/schema_1.1/mzML1.1.0.xsd for details. So scan 10 should look like this:

&lt;spectrum id="scan=10" index="9" defaultArrayLength="2"&gt;
&lt;cvParam accession="MS:1000511" value="1" name="ms level"/&gt;&lt;cvParam cvRef="MS" accession="MS:1000579" name="MS1 spectrum" value=""/&gt;
&lt;cvParam accession="MS:1000127" name="centroid spectrum"/&gt;
&lt;scanList&gt;
&lt;scan&gt;
&lt;cvParam accession="MS:1000016" value="308635" name="scan start time" unitAccession="UO:0000028" unitName="millisecond"/&gt;
&lt;/scan&gt;
&lt;/scanList&gt;
&lt;binaryDataArrayList&gt;
&lt;binaryDataArray encodedLength="28"&gt;
&lt;cvParam accession="MS:1000574" name="zlib compression"/&gt;
&lt;cvParam accession="MS:1000523" name="64-bit float"/&gt;
&lt;cvParam accession="MS:1000514" name="m/z array"/&gt;
&lt;binary&gt;eJxjYACChiiHNBBwi3IAABkOA/k=&lt;/binary&gt;
&lt;/binaryDataArray&gt;
&lt;binaryDataArray encodedLength="24"&gt;
&lt;cvParam accession="MS:1000574" name="zlib compression"/&gt;
&lt;cvParam accession="MS:1000521" name="32-bit float"/&gt;
&lt;cvParam accession="MS:1000515" name="intensity array"/&gt;
&lt;binary&gt;eJxjaOh0ZmBwcAYACNAB0A==&lt;/binary&gt;
&lt;/binaryDataArray&gt;
&lt;/binaryDataArrayList&gt;
&lt;/spectrum&gt;