OpenChrom/openchrom #213: Problem when importing GC/MS data

periodicsystem, 2022-02-18:
Hello, 
I am using OpenChrome/Eclipse on macOS. However, I have problems opening files (GC/MS). I can't drag and drop them into the interface or import them via system path import.
The only thing I can see is this surface. Unfortunately, I can't select anything or the like. When I click on Data Analysis, nothing happens either.

I would be very happy if anyone has any tips or a solution to this problem.

Thanks a lot
[image]
!

eselmeister, 2022-02-18:
Yep, I sent you a reply already to provide a file for testing purposes.

nhadler, 2022-03-04:
I have also been having this issue.

eselmeister, 2022-03-04:
@nhadler Could you send me some files for testing purposes?

ethanbass, 2022-03-26:
I think this is possible some kind of incompatibility with the m1 architecture rather than an issue with the types of files. Openchrom works fine for me on my old intel macbook, but doesn't run in Rosetta on my new m1 mac. Not sure if the original poster is also on an m1 mac.

Mailaender, 2022-03-27:
Check the latest version 1.5.0 on https://lablicate.com/platform/openchrom/download which we internally tested on M1 (with Rosetta). We have ARM builds on the agenda, but there is no ETA regarding them. However, we fixed the welcome screen bug. Sorry that it took us that long.

closed 2022-03-27 by Mailaender

ethanbass, 2022-03-27:
Success! Thank you. FIY, I had to give the app special permission to execute ```sudo xattr -cr /Users/ethanbass/Downloads/Eclipse.app```. It was giving me the dreaded "damaged file error"

Mailaender, 2022-03-27:
Yes, the executables are not signed, so they end in Gatekeeper's quarantine. There are plans to fix this soonish.