OpenChrom/openchrom #317: Crashing on linix(ubuntu)

pramit57, 2022-11-15:
I originally opened a thread on the adoptium support page : https://github.com/adoptium/adoptium-support/issues/628

The software crashes when I try to do almost anything from the starting screen. More details, including log files are in the thread.

Mailaender, 2022-11-15:
We will update Temurin with the upcoming release ETA on Friday.

Mailaender, 2023-10-26:
I doubt just updating the JVM helped. Upon investigation, I think this is https://github.com/eclipse-platform/eclipse.platform.swt/issues/678. It crashes because the icons in the tree viewer are updated. So it happens almost anywhere because most perspective have a Data Explorer part. For me, it often happens, but not always. When I remove all labelling, it seems to be gone. It seems to be also related to the GTK theme in use. I use @KDE's default Breeze on Kubuntu 23.04.

Mailaender, 2023-11-22:
I still get this crash consistently in a @flatpak environment.

Mailaender, 2023-11-22:
```
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f06aaeda0a1, pid=22, tid=23
#
# JRE version: OpenJDK Runtime Environment Temurin-17.0.9+9 (17.0.9+9) (build 17.0.9+9)
# Java VM: OpenJDK 64-Bit Server VM Temurin-17.0.9+9 (17.0.9+9, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)
# Problematic frame:
# C  [libgobject-2.0.so.0+0x3e0a1]  g_type_check_instance_is_fundamentally_a+0x11
```

tried both the 22.08 and 23.08 @freedesktop runtime.

Mailaender, 2023-12-01:
To reproduce, you have to delete your workspace. Then it always happens.

Mailaender, 2023-12-16:
Still crashes in a flatpak environment

```
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f7527edb0a1, pid=22, tid=23
#
# JRE version: OpenJDK Runtime Environment Temurin-17.0.9+9 (17.0.9+9) (build 17.0.9+9)
# Java VM: OpenJDK 64-Bit Server VM Temurin-17.0.9+9 (17.0.9+9, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)
# Problematic frame:
# C  [libgobject-2.0.so.0+0x3e0a1]  g_type_check_instance_is_fundamentally_a+0x11
```

Mailaender, 2024-02-01:
I finally have a workaround: https://github.com/eclipse/chemclipse/pull/1624

closed 2024-02-01 by Mailaender