OpenChrom/openchrom #347: Mol Weight Calculator - also calculate the formula

eselmeister, 2023-02-24:
Calculate additionally the formula if absent.

eselmeister, 2023-02-24:
Example Styrene:

[image]
[image]

Example Benzene:

[image]
[image]

closed 2023-02-24 by eselmeister

Mailaender, 2023-02-24:
Class and function should have been renamed.

commit 7be9d052 (2023-02-24): Fixed #347 - Mol Weight Calculator - also calculate the formula