OpenChrom/openchrom #402: GC-FID data: automatinc peak detector / integrator not working

AnnaClo, 2023-07-30:
I have data detected simultneously on gc ms and fid. OpenChrom detects and integrates properly the ms data, but not the fid peaks. For some reason some of the peaks result integrated only halfaway - see screenshots. 
Any idea what could be the fix? 

Here below a peak detected and integrated with the same procedure, from ms data, and from fid data respectively.

[image]

[image]

eselmeister, 2023-07-31:
With GC-FID, smoothing might be required first. Could you provide the displayed data set?

AnnaClo, 2023-07-31:
Yes sure, here is a link to the same dataset. The peaks are very tiny compared to the solvent, between 13 and 27 mins. https://we.tl/t-AfNfUjLLBc

eselmeister, 2023-07-31:
Thanks for the file. I would recommend to run the following steps on the GC-FID chromatogram:

1) Remove the solvent and first big peak via Chromatogram Filter > Scan Remover
2) Set a baseline via Baseline Detector > SNIP (Iterations: 250, Size: 9)
3) Remove the baseline via Chromatogram Filter > Chromatogram Baseline Subtract Filter
4) Smooth the data Chromatogram Filter > Savitzgy-Golay (Order: 2, Width: 9)
5) Detect peak via Peak Detector > First Derivative (Include Background: yes, Use Noise Segments: yes, Optimize Baseline: yes)

[image]

closed 2023-08-01 by eselmeister