OpenChrom/openchrom #403: Opening projects and batch jobs (on different computers)

AnnaClo, 2023-08-02:
Hi, I'm struggling to work with project and batch jobs. It's no problem to create a project and a batch job, in a folder that is different from default. However I don't understand how to open an existing project or batch job. I tried importing but I don't understand the dialogue box

eselmeister, 2023-08-03:
@AnnaClo Use the "Project Explorer". You can see the location via pop-up menu right mouse click in the settings. Additionally, you can link any folder from your system.

[image]
[image]
[image]
[image]

AnnaClo, 2023-08-03:
Yes, I created a project in a folder in my system (on cloud), but I'm stuck when I try to import that project in OpenChrom (on a different computer that also connects to the same cloud folder : 
[image]
[image]
[image]
I would like also to import the batch job that I create within that project if possible

[image]
[image]

eselmeister, 2023-08-04:
First of all, create a simple project. Than copy & paste your batch file (e.g. BatchJob1.obj) into the new project folder:

[image]
[image]
[image]
[image]

closed 2023-08-14 by eselmeister