OpenChrom/openchrom #411: Question about peak identifier

AnnaClo, 2023-08-13:
I need to identify peaks based on the retention time. I did so for a batch of saples via the Peak identifier > Peak identifier [template]

The problem is that the RT shifts gradually, so that means I need to edit manually the RTs in the template for every batch of ca. 60 samples.
 
Is there a more convenient way to adjust / shift the RT of the peak identifier? 

[image]

[image]

eselmeister, 2023-08-14:
@AnnaClo The last screenshot "Peak Identifier [Template]" offers an option to shift retention times batchwise. It's the icon right beside the search symbol. When activated, new options appear to shift the retention times. It offers 6 options

- Retention Time Minutes start/stop
- Retention Time Milliseconds start/stop
- Retention Inded start/stop

See column "Position Directive" what you need. Additionally, I recommend to use "Retention Indices". Then the templates can be used almost without modification. But you need to run alkane standards and calculate the RI in your chromatograms.

AnnaClo, 2023-08-14:
Ah I see, that's already very helpful!
I tried indeed to use RI, however I'm not working with alkanes. I'm working with PLFAs, in that case I can calculate ECL values based on the time of the RT C12:0 fatty acid and the C19:0 fatty acid. The ECL - conceptually similar to RI, is stable across batches.
Could there be a way to calculate ECL/RI within openchrom based on anything else than alkanes?

eselmeister, 2023-08-14:
That should do the trick. Simply modify the *.cal text file (can't upload, hence content is printed) and load it into the `Chromatogram Calculator > Retention Index Calculator (embedded)`.

```
#COLUMN_NAME=DEFAULT
#COLUMN_LENGTH=
#COLUMN_DIAMETER=
#COLUMN_PHASE=
1.908 600.0 100 999 C6 (Hexane)
2.917 700.0 100 999 C7 (Heptane)
4.258 800.0 100 999 C8 (Octane)
5.783 900.0 100 999 C9 (Nonane)
7.333 1000.0 100 999 C10 (Decane)
8.85 1100.0 100 999 C11 (Undecane)
10.3 1200.0 100 999 C12 (Dodecane)
11.675 1300.0 100 999 C13 (Tridecane)
12.983 1400.0 100 999 C14 (Tetradecane)
14.225 1500.0 100 999 C15 (Pentadecane)
15.408 1600.0 100 999 C16 (Hexadecane)
16.525 1700.0 100 999 C17 (Heptadecane)
17.6 1800.0 100 999 C18 (Octadecane)
18.625 1900.0 100 999 C19 (Nonadecane)
19.608 2000.0 100 999 C20 (Eicosane)
20.55 2100.0 100 999 C21 (Heneicosane)
21.458 2200.0 100 999 C22 (Docosane)
22.333 2300.0 100 999 C23 (Tricosane)
23.167 2400.0 100 999 C24 (Tetracosane)
23.975 2500.0 100 999 C25 (Pentacosane)
24.758 2600.0 100 999 C26 (Hexacosane)
25.508 2700.0 100 999 C27 (Heptacosane)
26.242 2800.0 100 999 C28 (Octacosane)
26.942 2900.0 100 999 C29 (Nonacosane)
27.633 3000.0 100 999 C30 (Triacontane)
28.35 3100.0 100 999 C31 (Hentriacontane)
29.175 3200.0 100 999 C32 (Dotriacontane)
```

[image]
[image]
[image]

AnnaClo, 2023-08-14:
Ok, I modified the file in Chromatogram Calculator > Retention Index Calculator (embedded) and saved it - it was saved as .rim file though, I didn't see the option of having a .cal
[image]
Then I see the ladder overlayed to the chromatogram. 

[image]

Since I didn't run alkanes, I was looking for a way of calculated the RI of peaks in one of the fatty acids standards. I did Chromatogram export > AMIDS RI (.CAL) and then did  Chromatogram Calculator > Retention Index Calculator (Scans and Peaks) and used the .CAL obtained from that standard. In this way I get a list of RIs for each of my fatty acids. 
Then I could use a template for peak identification with those RIs +- a few units. 
Is there anything weird in this process?

eselmeister, 2023-08-15:
@AnnaClo Correct, you can import/export the entries as *.rim file. There's a button right beside the green plus button to import an AMDIS *.cal file.

The alkane raster looks good. Your process looks fine. You don't need to modify the entries in your Retention Index Calculator  each and every time. They are saved, respectively you should see them listed. Yes, it can be used in a process method and thus also in a batch process. The only issue left is, that you have to change you template method from "Position Directive" -> Minutes to Retention Index. You can export the template and do it also in a text editor like Notepad++.

Good luck with your analysis.

closed 2023-08-15 by eselmeister