OpenChrom/openchrom #435: OpenChrom Flatpak package

leArthurDent, 2023-11-20:
Hi there,

is there any possibility to provide OpenChrom as a [Flatpak](https://www.flatpak.org/) package for linux?
Several linux distributions such as Fedora Silberblue and OpenSUSE ALP are switching to immutable operating systems were apps can merely be installes via Flatpak.

I know you have a snap package, but the snap daemon is quite bloated and snap is merely used by ubuntu and its official flavours.

Cheers.

eselmeister, 2023-11-22:
@leArthurDent Thanks for your request. We will have a look at it.

Mailaender, 2023-11-22:
I created one a few months ago, but it is blocked by https://github.com/OpenChrom/openchrom/issues/317.

closed 2024-05-06 by Mailaender

Mailaender, 2024-05-08:
https://flathub.org/apps/com.lablicate.OpenChrom