OpenChrom/openchrom #437: Question about .raw file conversion

AnnaClo, 2023-11-27:
Hi, I have a question about MSD .RAW files obtained from a Thermo Fisher GC-MS. 
These .raw files are very large and difficult to work with. We tried running a (batch) method and also converting the .raw file into .ocb (export chromatogram > Open chromatography binary .ocb),  but both processes take a very long time. 
Is there a way to bypass this problem, e.g. another way for converting the files into a smaller format ?

eselmeister, 2023-11-28:
@AnnaClo Could you provide a few chromatograms for testing purposes?

AnnaClo, 2023-11-28:
Yes sure, I send here one injection, as a we transfer because the file is too large

https://we.tl/t-Jn1PGpPzfp

eselmeister, 2023-11-29:
Do you expect that your chromatogram looks like this?

[image]

I was able to pre-process it and reduce the size from 1.3 GB down to 7.4 MB. It can be done with the process methods.

[image]

Condensed Chromatogram:
[BAME_std1A 1 v1.zip](https://github.com/OpenChrom/openchrom/files/13499463/BAME_std1A.1.v1.zip)

AnnaClo, 2023-11-29:
Yes, that's how I expect the chromatogram to look like. 
In the v1 file (.ocb) appears an empty chromatogram to me right now. This is something that happened with more files, after processing the raw file, and saving the result as ocb, that turns out to be empty.

The problem we have is that the preprocessing itself takes very long (e.g. baseline subtraction filter takes already a lot of time). Is there a preprocessing step that allows to reduce the size of the file that we can use first of all other ones?  

[image]

eselmeister, 2023-11-29:
Do you use the latest version of OpenChrom? Could you share the version number: Menu > Help > About

AnnaClo, 2023-11-29:
Here it is:  version 1.5.0
[image]

eselmeister, 2023-11-29:
All right, we have updated the *.ocb format recently:
https://github.com/eclipse/chemclipse/issues/1326

The *.ocb version I have uploaded uses the latest version 1.5.0.1 (McLafferty v2). Could you try to open it with the latest OpenChrom release:
https://openchrom.net/download

AnnaClo, 2024-01-30:
Ok, thank you for helping. Indeed, we managed it by letting the batch process run for several hours.

eselmeister, 2024-01-30:
You're welcome.

closed 2024-01-30 by eselmeister