OpenChrom/openchrom #452: could not load and convert sciex wiff format

amenglar, 2024-02-02:
hi, I'm trying to find a tool on Linux OS to convert vendor ms files (including sciex, waters, etc) to mzml format. the most popular solution ms-convert is only available on windows. the openchrom with the ability to convert and run on linux is a good choice.

however when I firstly tested a chromatogram sample of sciex wiff format, it was nothing loaded. 
I also tested this file with msconvert (proteowizard), I could successfully load the peaks.

[image]
[DataYZ12-20240110-AD-S1-V3.wiff.zip](https://github.com/OpenChrom/openchrom/files/14133413/DataYZ12-20240110-AD-S1-V3.wiff.zip)

eselmeister, 2024-02-03:
@amenglar Thanks for the demo file. We will put it on our todo list.

amenglar, 2024-02-06:
Another problem, I couldn't download of the older version, 1.4.x by either way. I don't know if I can post it here.

[image]

eselmeister, 2024-02-06:
Yep, no worries. We recommend to use the latest 1.5 rolling release. Implementing the *.wiff converter is on our todo list.

ricoderks, 2025-03-04:
@amenglar Have a look at [this](https://hub.docker.com/r/chambm/pwiz-skyline-i-agree-to-the-vendor-licenses) page. Here msconvert runs in a docker container. I've been using this a lot to convert to mzxml / mzml on a Linux machine.