OpenChrom/openchrom #489: Import CDF with RetentionUnit="s" specified

avasyanin, 2024-03-28:
CDF files exported by Chromatec Analytic 2.6 program have RetentionUnit specified as "s", not "seconds", so their time scale after reading by OpenChrom is incorrect. The issue probably can be easily solved by adding "s" to the line 87 of https://github.com/OpenChrom/openchrom/tree/develop/openchrom/plugins/net.openchrom.csd.converter.supplier.cdf/src/net/openchrom/csd/converter/supplier/cdf/io/support/AbstractCDFChromatogramArrayReader.java file:

```
if(unit.equals("seconds") || unit.equals("Seconds")) {
```

Thanks.

eselmeister, 2024-03-28:
Great, thanks for the hint. Could you share a CDF file for testing purposes?

avasyanin, 2024-03-28:
Yes, sure.

The file also contains some supporting information which is in an ancient CP1251 (Cyrillics), but actually it is not very important.

[20240327-174018 [169.3] (1).cdf.zip](https://github.com/OpenChrom/openchrom/files/14790727/20240327-174018.169.3.1.cdf.zip)

eselmeister, 2024-03-30:
@avasyanin Thanks for the file. It has been fixed and will be shipped with the next release, scheduled for Friday, 5th of April.

[image]

commit 125f9831 (2024-03-30): Fixed #489 - Import CDF with RetentionUnit="s" specified
closed 2024-03-30 by eselmeister