OpenChrom/openchrom #514: Batch Process Job editor not appearing.

johnll56, 2024-07-04:
Currently trying to batch process some HPLC-DAD chromatograms from a Shimadzu instrument. The program reads them just fine.
However, when I go to Chromatogram -->Batch Process, create a new Project and inside it a new Batch Process Job, i get sent straight to the files and method selection window without seeing the editor. Therefore I cannot select an export folder and when clicking "Execute" something seems to happen but no output files are created.
Running Openchrom v1.5 McLafferty edition.

Please help
[image]

This is the screen I see immediately after creating the Batch Process Job and selecting WSD as my data type. No editor in sight

eselmeister, 2024-07-08:
@johnll56 Are you able to drag & drop the Shimadzu *.lcd files in the left part of the editor and create new process entries with the green plus button in the right part of the editor?

johnll56, 2024-07-08:
> @johnll56 Are you able to drag & drop the Shimadzu *.lcd files in the left part of the editor and create new process entries with the green plus button in the right part of the editor?

Yes, and I ended up processing each chromatogram individually by creating a method that made my required modifications and then exported the file. It would still be worth it to check the code because as the original post states, the batch process job editor never appeared. I even tried it in the Data Analysis section (after selecting one wavelength and converting one of my .lcd files to .ocb) and it didn't work.

eselmeister, 2024-07-09:
Could you provide some *.lcd files for testing purposes?

johnll56, 2024-07-09:
> Could you provide some *.lcd files for testing purposes?

Sure, here are a couple. Uploading as .zip file with no password as Github doesn't support .lcd attachments. Files named 1-3 are from the calibration curve I had to construct as part of the experiment and files 4-6 were obtained from actual biological samples.
[lcd chromatograms.zip](https://github.com/user-attachments/files/16143834/lcd.chromatograms.zip)

Thanks. Hopefully you can also reproduce the bug.

eselmeister, 2024-07-12:
@johnll56 The batch process works fine on my side. Keep in mind to select "WSD" as an option when creating the batch process and to install the Shimadzu *.lcd converter accordingly.

[image]
[image]
[image]

closed 2024-07-14 by eselmeister

johnll56, 2024-07-15:
Thanks for checking, might have been a Java bug on my end or something.