OpenChrom/openchrom #523: "Create Retention Index File" menu entry grayed out

concifederico, 2024-07-26:
Hi,
Can't make it enabled, no matters what I do. ver. 1.5.0.20220617

[image]

Any help will be appreciated.
Regards,

Federico

eselmeister, 2024-07-26:
@concifederico Please provide your raw file for demo purposes.

concifederico, 2024-08-01:
[Uploading std alkanos 140823.raw.zip…]()
Here we go!

eselmeister, 2024-08-05:
Thanks for the file. I can't reproduce it.
[image]

eselmeister, 2024-08-05:
If you're a student, get a free license to use our "Power Tools" to easily create RI *.cal files:
https://lablicate.com/marketplace/extensions/powertools

[image]
[image]

closed 2024-08-05 by eselmeister