OpenChrom/openchrom #530: Tandem MS analyses: Missing information about selected precursor ions

LouisPM, 2024-08-31:
Dear _OpenChrom_ developers,

_ChemClipse_ seems to support the import of _.mzML_ files originating from tandem MS analyses. And in _OpenChrom_, 'Filters' enable one to split tandem MS data into chromatograms and ion scans according the level of the MS analysis.

However, information about the precursor ion that was selected for fragmentation seems nowhere to be found. For example, consider an MS/MS analysis, where precursor ions (MS1) had been selected for fragmentation, and scans of their respective fragment ions (MS2) have been recorded. In order to make sense the MS2 spectra, ones needs the information about the respective precursor ion, most crucially its _m/z_ value.

I suggest to add the precursor _m/z_ to both, the window "Chromatogram Scan Info" (_e.g._ as an additional table column), and to the window "Scan Chart" in the line shown upon clicking on the option "Show additional information." In case this information is missing, one could default to al null value or NaN.

It appears that the precursor ion _m/z_ information is described within _.mzML_ headers. Here is an example excerpt for an MS2 scan:
`<precursorList count="1">
            <precursor spectrumRef="controllerType=0 controllerNumber=1 scan=1425">
              <isolationWindow>
                <cvParam cvRef="MS" accession="MS:1000827" value="387.686004638672" name="isolation window target m/z" unitAccession="MS:1000040" unitName="m/z" unitCvRef="MS" />
                <cvParam cvRef="MS" accession="MS:1000828" value="0.8" name="isolation window lower offset" unitAccession="MS:1000040" unitName="m/z" unitCvRef="MS" />
                <cvParam cvRef="MS" accession="MS:1000829" value="0.8" name="isolation window upper offset" unitAccession="MS:1000040" unitName="m/z" unitCvRef="MS" />
              </isolationWindow>
              <selectedIonList count="1">
                <selectedIon>
                  <cvParam **cvRef="MS"** accession="MS:1000744" **value="387.686004638672" name="selected ion m/z"** unitAccession="MS:1000040" unitName="m/z" unitCvRef="MS" />
                  <cvParam cvRef="MS" accession="MS:1000042" value="9859.3564453125" name="peak intensity" unitAccession="MS:1000131" unitName="number of detector counts" unitCvRef="MS" />
                </selectedIon>
              </selectedIonList>
              <activation>
                <cvParam cvRef="MS" accession="MS:1000045" value="30" name="collision energy" unitAccession="UO:0000266" unitName="electronvolt" unitCvRef="UO" />
                <cvParam cvRef="MS" accession="MS:1000422" value="" name="beam-type collision-induced dissociation" />
              </activation>
            </precursor>
</precursorList>`

Mailaender, 2024-10-15:
Can you share an example file?

LouisPM, 2024-10-16:
The files I have at hand would be to large for an upload on _GitHub_.
Instead, I sent you an invite to a private repository containing a download link.
Let me know once you managed to download the file.

Mailaender, 2024-10-16:
I could download them.

closed 2024-10-16 by eselmeister